SasaKaranovic/DIY-Filament-Dryer #5: PID values when 'Using HEATER as target temperature' returning negative

benuron, 2023-10-02:
Hi Sasa, thanks for sharing this project and make it available to everyone :)

I am trying to setup the whole project since a while now, I am not a code expert so my efforts to fix some bugs are reaching a limit.

One situation that is that when the max_heater_temperature reaches the set value the PID values starts to get negative and the calculated PWM values shoot to the max and the heater keeps ON. 

[image]

Am I missing something?
I am trying to create a branch and contribute for this project as soon as I understand git better 👍

SasaKaranovic, 2024-02-29:
@benuron appologies for late reply. I somehow missed that you have opened this issue.

Did you modify the code or using it as-is?

I am working on a new version of this project that should hopefully be better and offer some quality of life improvements.

benuron, 2024-03-03:
@SasaKaranovic thanks for replying!

I am using some slightly modified code but nothing regarding the PID. In my code I changed the wifi connection, and I am using two SHT31 ```
Adafruit_SHT31 temp_sensor_in = Adafruit_SHT31();
Adafruit_Si7021 temp_sensor_out = Adafruit_SHT31();
```
and in this section I modified the i2c address in the temp_sensor_out
 
```
 // Setup temperature and humidity oneWireSensors
    if (!temp_sensor_out.begin(0x45))
    {
        Serial.println("Can't find Si7021 sensor! (out sensor)");
        LED_ALL_OFF();
        while (1)
        {
            LED_ON(PIN_LED_BLUE);
            delay(250);
            LED_OFF(PIN_LED_BLUE);
            delay(250);
        }
    }

    Wire.setClock(10000);
    if (!temp_sensor_in.begin(0x44))
    {
        Serial.println("Can't find SHT31! (in sensor)");
        LED_ALL_OFF();
        while (1)
        {
            LED_ON(PIN_LED_GREEN);
            delay(250);
            LED_OFF(PIN_LED_GREEN);
            delay(250);
        }
    }   if (!temp_sensor_out.begin())
    {
        Serial.println("Can't find Si7021 sensor! (out sensor)");
        LED_ALL_OFF();
        while (1)
        {
            LED_ON(PIN_LED_BLUE);
            delay(250);
            LED_OFF(PIN_LED_BLUE);
            delay(250);
        }
    }

    Wire.setClock(10000);
    if (!temp_sensor_in.begin(0x44))
    {
        Serial.println("Can't find SHT31! (in sensor)");
        LED_ALL_OFF();
        while (1)
        {
            LED_ON(PIN_LED_GREEN);
            delay(250);
            LED_OFF(PIN_LED_GREEN);
            delay(250);
        }
    }
```
From the webUI, all values are good, just the strange behaviour with the PID. I tried to manually tweak them a bit but without much success. Would be possible to have a PID autotune feature like in Marlin firmware?

There is a difference in hardware from my S1 (I have five) to yours. They all run with 24V power supplies. With the BOM for 12V the MOSFET generates a bunch of heat. I had to recalculate the voltage divider values so the MOSFET is full on/off. 

Would love to try the new version, thank you for the efforts!

SasaKaranovic, 2024-06-05:
Closing due to inactivity. We can pick it up if new information becomes available.

closed 2024-06-05 by SasaKaranovic

razserv2010, 2024-08-01:
> @benuron appologies for late reply. I somehow missed that you have opened this issue.
> 
> Did you modify the code or using it as-is?
> 
> I am working on a new version of this project that should hopefully be better and offer some quality of life improvements.

Hi SasaKaranovic
Is there already a new version?