SasaKaranovic/DIY-Filament-Dryer #6: A fatal error occurred: Failed to connect to ESP32: No serial data received.

razserv2010, 2024-02-19:
I'm experiencing a error:
A fatal error occurred: Failed to connect to ESP32: No serial data received.
I connected using FTDI232
3V3 - 3V3
TX - RX
RX - TX
GND - GND

[image]
[image]

SasaKaranovic, 2024-02-19:
Hi Raz,

Sorry about that, there was a typo in the BoM but it's fixed now.
You would need `MIC5365-3.3YD5-TR` (or any other equivalent 3v3 LDO).

razserv2010, 2024-02-19:
> Hi Raz,
> 
> Sorry about that, there was a typo in the BoM but it's fixed now.
> You would need `MIC5365-3.3YD5-TR` (or any other equivalent 3v3 LDO).

So this is the reason why the pc does not recognize the esp32?

SasaKaranovic, 2024-02-19:
Yes. The regulator you are using is providing ESP32 with 1.8V which is not enough for it to power up.
If you switch to a 3.3V regulator, it should work (assuming everything else is ok).

razserv2010, 2024-02-19:
> Yes. The regulator you are using is providing ESP32 with 1.8V which is not enough for it to power up.
> If you switch to a 3.3V regulator, it should work (assuming everything else is ok).

Thanks, I ordered 3v

SasaKaranovic, 2024-02-19:
Sorry about that. The release files and the Altium files were updated a long time ago but when you mentioned it, I realized that the schematic is still calling out the wrong LDO.

razserv2010, 2024-02-19:
> Sorry about that. The release files and the Altium files were updated a long time ago but when you mentioned it, I realized that the schematic is still calling out the wrong LDO.

Sorry but I didn't understand, LDO?

SasaKaranovic, 2024-02-20:
[LDO regulator](https://en.wikipedia.org/wiki/Low-dropout_regulator).
The schematic had the wrong LDO part number. If you replace it with 3.3V regulator, it should work.

razserv2010, 2024-02-20:
> [LDO regulator](https://en.wikipedia.org/wiki/Low-dropout_regulator).
> The schematic had the wrong LDO part number. If you replace it with 3.3V regulator, it should work.

Thanks, I ordered 3.3V

razserv2010, 2024-02-25:
> [LDO regulator](https://en.wikipedia.org/wiki/Low-dropout_regulator). The schematic had the wrong LDO part number. If you replace it with 3.3V regulator, it should work.

I replaced the component but still:
A fatal error occurred: Failed to connect to ESP32: No serial data received.

What is the order of UART connections? , maybe I didn't connect correctly..

razserv2010, 2024-02-26:
> [LDO regulator](https://en.wikipedia.org/wiki/Low-dropout_regulator).
> The schematic had the wrong LDO part number. If you replace it with 3.3V regulator, it should work.

 After I replaced the component, I was able to upload the firmware but the esp32 does not connect to wifi

razserv2010, 2024-02-26:
Until the Si7021 sensors arrive, I want to connect the board but it won't let me connect to wifi
How can Si7021 be bypassed?

katodo, 2024-02-27:
you can start to bypass "if (!temp_sensor_out.begin())" at the begin of code.

SasaKaranovic, 2024-02-29:
@razserv2010 As @katodo said you could stub out (comment out) parts of the code that rely on Si7021.
But it might be tedious and also I would recommend using the sensor since some portions of the code depend on it.

I am working on new version of this board that should offer some improvements that people have asked for.

razserv2010, 2024-03-01:
> @razserv2010 As @katodo said you could stub out (comment out) parts of the code that rely on Si7021.
> But it might be tedious and also I would recommend using the sensor since some portions of the [...]

The sensors arrived, trying to connect to wifi but without success, I opened a new issue about it.

closed 2024-03-02 by SasaKaranovic