SasaKaranovic/DIY-Filament-Dryer #11: errors in compile wiith platformio

Motarbio, 2024-06-05:
compiling aborted with this errors

> pio run
Processing esp32dev (framework: arduino; platform: espressif32; board: esp32dev)
---------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.7.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)      
PACKAGES:
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-openocd-esp32 @ 2.1100.20220706 (11.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Converting src.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- FS @ 2.0.0
|-- WiFi @ 2.0.0
|-- Adafruit SHT31 Library @ 2.2.2
|-- Adafruit Si7021 Library @ 1.5.3
|-- ESP Async WebServer @ 1.2.3
|-- AsyncTCP @ 1.1.1
|-- DallasTemperature @ 3.11.0
|-- OneWire @ 2.3.8
|-- ESPmDNS @ 2.0.0
|-- Wire @ 2.0.0
Building in release mode
Compiling .pio\build\esp32dev\src\src.ino.cpp.o
Linking .pio\build\esp32dev\firmware.elf
c:/users/magut/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib0ed\libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x4): undefined reference to `mbedtls_md5_starts'       
c:/users/magut/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib0ed\libESP Async WebServer.a(WebAuthentication.cpp.o): in function `getMD5(unsigned char*, unsigned short, char*)':
C:\Users\Magut\Downloads\DIY-Filament-Dryer-master\DIY-Filament-Dryer-master\Firmware\platformio/.pio/libdeps/esp32dev/ESP Async WebServer/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1

SasaKaranovic, 2024-06-05:
Hey @Motarbio 

From the output it looks like PlatformIO is identifying conflict between libraries.

Can you try updating `platformio.ini` file and changing lines Line 29 and Line 30 (https://github.com/SasaKaranovic/DIY-Filament-Dryer/blob/master/Firmware/platformio/platformio.ini#L29-L30) to be as follows

```
    https://github.com/me-no-dev/ESPAsyncWebServer.git
    https://github.com/me-no-dev/AsyncTCP.git
```

Motarbio, 2024-06-05:
Hi Now the error is please install  git clientDegiorgi Tiziano

    https://github.com/me-no-dev/ESPAsyncWebServer.git
    https://github.com/me-no-dev/AsyncTCP.git

SasaKaranovic, 2024-06-05:
@Motarbio Can you try doing that?
You can install a Git client from https://git-scm.com/download/win (assuming you use Windows)

Motarbio, 2024-06-05:
Done, nothing dame errorDegiorgi Tiziano

SasaKaranovic, 2024-06-05:
Please try restarting your computer after installing Git client and then try again.

p.s. When you reply via email, it messes up the formatting on the GitHub issue.

Motarbio, 2024-06-05:
pc restarted, same error
Library Manager: Installing git+https://github.com/me-no-dev/ESPAsyncWebServer.git
UserSideException: Please install Git client from https://git-scm.com/downloads

Motarbio, 2024-06-05:
Hi there,
downloaded manually the libraries, compiled succesfully but with 2 warnings

Library Manager: Installing ESPAsyncWebServer-master
Warning! Could not find the package with 'ESPAsyncWebServer-master' requirements for your system 'windows_amd64'
Library Manager: Installing AsyncTCP-master
Warning! Could not find the package with 'AsyncTCP-master' requirements for your system 'windows_amd64'

SasaKaranovic, 2024-06-06:
You should double check if `git` is in your system `PATH` variable. Then PlatformIO should be able to pull those libraries for you automatically.