SasaKaranovic/DIY-Filament-Dryer #12: ignores heater temp limit

gabecamp, 2024-06-15:
seems to ignore the heater temp limit and goes to 100c ,built as normal except for using 2 Adafruit SHT31 as in and out sensors,

SasaKaranovic, 2024-06-15:
@gabecamp Is the heater sensor reporting that temperature or you manually measured it?
Can you share which components you used and which PCB vendor?

gabecamp, 2024-06-16:
Using all the parts from the BOM on GitHub except for 2 sh31 sensors , the PCBs were using the Gerber's from github with jlcpcb. The temps are from the ds18b20 as per the schematic.  After some testing I found it does hold temp correctly after turning it off and letting it cool a few times. Could it possibly be the pid loop causing thermal runaway?

SasaKaranovic, 2024-06-17:
It's good that it's stable now.
I had mine running for couple of years now and had no issues.
If you find way to reproduce this, please let me know.